Fabi0San/DuCalibrator #1: Klipper support

apprentice10, 2020-06-24:
I'll follow this because is interesting, but I'm using klipper.

Fabi0San, 2020-06-24:
Absolutely.
I plan to switch my printers to Klipper at which point this will be easy. 
Until then, if anyone running Klipper with some dev'ing experience has the time to help out, contact me and we can make it happen.

rbrian, 2020-06-26:
get in touch. i can help with klipper integration, just need some pointers as to how. email rblindahl@gmail.com.

Fabi0San, 2020-06-26:
We can talk about it here so others can contribute:

My initial questions are:
What are the Klipper equivalent for these commands:

            Home: "G28",
            Echo: "M118",
            Move: "G0",
            ProbeBed: "G30",
            FetchSettings: "M503",
            SaveSettings: "M500",
            StepsPerUnit: "M92",
            EndStopOffset: "M666",
            DeltaConfig: "M665",

What are the identifiers for these geometry parameters:

            idsRadiusHeightRod: "RHL",
            idsTowerAngleOffset: "XYZ",
            idsEndStopOffset: "XYZ",         
            idsStepsPerUnit: "XYZ",
            DiagonalRodAdjust: "MNO",

I also need to see the output of M503 and G30, 
also is the G30 Z measurement the absolute machine position at bed touch like Marlin or the relative distance from where the measurement started like Smoothie?

Thanks for your help.

apprentice10, 2020-06-26:
>         Home: "G28",
Same "G28"
```
Send: G28
Recv: ok
```
___________________________________________________________________________________________________________
>         Echo: "M118",
You need to activate the "respond" section on the config file [https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L513-L523](url)
For other commands here: [](https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#send-message-respond-to-host)

```
Send: M118 NEED MORE POWER!!!
Recv: echo: NEED MORE POWER!!!
Recv: ok
```
___________________________________________________________________________________________________________
> Move: "G0",
Same "G0"
```
Send: G0 X50 Y50 Z10
Recv: ok
```
___________________________________________________________________________________________________________

> ProbeBed: "G30",
You need to activate the "Probe" hardware section to use this command (If you have a working probe you already done this)
For my understanding, you need to move the head to the XY position first (and a safe but close distance from the bed to for faster movements) and send the PROBE command.
```
Send: G28
Recv: ok
Send: G0 X50 Y50 Z10
Recv: ok
[...]
Send: PROBE
Recv: // probe at 49.998,50.005 is z=-0.438369
Recv: // Result is z=-0.438369
Recv: ok
```
___________________________________________________________________________________________________________

> FetchSettings: "M503",
> SaveSettings: "M500",

I've made some research and apparently there's no such GCODE mostly because KLIPPER doesn't save the settings on the EEPROM, instead it uses a printer.cfg file in this path: "/home/pi/printer.cfg"

[printer cfg.zip](https://github.com/Fabi0San/DuCalibrator/files/4836600/printer.cfg.zip)
I don't know how hard it is to fetch and change that file.
___________________________________________________________________________________________________________

> StepsPerUnit: "M92",

Same as M503 & M500. In Klipper printer.cfg file is "step_distance" instead of "StepsPerUnit": from Marlin you have to make this calculation: Klipperstep_distance= 1 / MarlinStepsPerUnit
___________________________________________________________________________________________________________

> EndStopOffset: "M666"
Same as M503 & M500. In Klipper printer.cfg file is "position_endstop" under each stepper section:

```
[stepper_a]
#position_endstop: 295.092444
#   Distance (in mm) between the nozzle and the bed when the nozzle is
#   in the center of the build area and the endstop triggers. This
#   parameter must be provided for stepper_a; for stepper_b and
#   stepper_c this parameter defaults to the value specified for
#   stepper_a.
```

If you run the DELTA_CALIBRATE command it will comment the default line and add a new one at the bottom of the file:
```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# [stepper_a]
#*# angle = 209.258349
#*# arm_length = 265.065367
#*# position_endstop = 299.527025

```

> DeltaConfig: "M665"
Same as M503 & M500.

... The more I look into this the less liketly is see it fitting for Klipper.

Fabi0San, 2020-06-26:
This is very helpful, thank you very much @apprentice10 
Don't despair, it will be a bit trickier but we will find a way. 
From what I see we need a gcode way to fetch/set delta geometry. 
Maybe @KevinOConnor can tell us if that is really not already done, nor easily achievable.

As last resort we can always go via Octoprint(instead of gcode) to either interface with Klipper plugin(rest?, viewmodel?) or get to the config files directly, though that is more fragile due to future changes to Klipper configuration schema, etc.

KevinOConnor, 2020-06-27:
FYI, there is currently no mechanism to update Klipper's delta parameters from an external program.  Note, however, that Klipper does have an internal delta calibration tool which can probe, calculate, and then update Klipper's config file - more info at: https://www.klipper3d.org/Delta_Calibrate.html

-Kevin

apprentice10, 2020-06-27:
Yes! And I can't complain. It works great. But I really liked the interface in this plugin.

rbrian, 2020-08-05:
I like two things about this plugin: one, it repeats the probing until it gets a consistent value (like Marlin's delta config)... and it gives a nice graphical representation of the probed points. I'm having issues with what appears to be bed glass that isn't even reasonably flat, much less perfectly flat. Having a 3d view of that would be very helpful.