Fabi0San/DuCalibrator #2: Cannot Fetch Geometry

snown, 2020-06-24:
I'm unsure of what I'm missing in the set up of this plug-in. I think I entered all of the settings correctly, but the plug-in seems to just spin and not allow me to interact with the 3rd tab. This happens on a simulated printer too.

Here is a video of what I'm seeing:
https://www.dropbox.com/s/gm4wdenzs5398df/DuCalibrator.mp4?dl=0

Any advise would be appreciated. Thank you.

Fabi0San, 2020-06-24:
Quite the failure, maybe I should add checks for that.
Which browser and version are you using?
My guess is that your browser are not compatible to all the spunky new features I'm trying to use there, which includes webgl and a lot of ES6 javascript language features.
If you can, check the browser's console messages, it may give us some clues to what is not being loaded right.

Danbam, 2020-06-25:
I am having this same issue. Have tried using safari and chrome.

BrunoBrutus, 2020-06-25:
Hello @Fabi0San 

First of all, I got very excited to see that you here able to apply the tool from David Crocker, on Octopi.

Unfortunately I'm also getting the same problem, already tried Chrome (V83 64bits), Firefox, Edge and Internet explorer 6.
My system configuration below:
->Raspberry Pi 2 Model B Rev 1.1
->octopi_version: 0.16.0
->Kossel Plus with a Trigorilla 1.4 board with Marlin 1.1.9.1, pre aligned and ready to print.

But I started logging my Octopi and got the following warning on the "octoprint.log":
*octoprint.server.util.flask.collect_plugin_assets - WARNING - Plugin DuCalibrator is referring to non existing js asset js/DucalCommon.js*

Can you point us to on how to get this asset?

On Chrome I started the developers tool and got the below errors:
*packed_plugins.js?fa5c24c6:54078 Error in JS assets for plugin DuCalibrator: ReferenceError: AlphaTower is not defined*

*Error instantiating duCalibratorViewModel : ReferenceError: CollapseControl is not defined
    at new DuCalibratorViewModel*

Got a question, is it normal to insert all the required data (print radius, probe points, initial commands, probe distance) every time the system is rebooted or refreshed?

Hope you have enough information to debug :)

Thank you for your support.

Fabi0San, 2020-06-25:
>First of all, I got very excited to see that you here able to apply the tool from David Crocker, on Octopi.

Well, we both use least squares linear interpolation for finding where the factor converge but this is far from David's tool ;)

>But I started logging my Octopi and got the following warning on the "octoprint.log":
octoprint.server.util.flask.collect_plugin_assets - WARNING - Plugin DuCalibrator is referring to non existing js asset js/DucalCommon.js

Oh boy, this is really embarrassing, I tested on windows, messed up the casing of the asset file name declaration. 

>On Chrome I started the developers tool and got the below errors:
packed_plugins.js?fa5c24c6:54078 Error in JS assets for plugin DuCalibrator: ReferenceError: AlphaTower is not defined

It is defined in DuCalCommon.js...

> Got a question, is it normal to insert all the required data (print radius, probe points, initial commands, probe distance) every time the system is rebooted or refreshed?

radius and points yes, commands and distance should be saved when you click save settings on the Octopi UI.

>Hope you have enough information to debug :)

More than enough thank you very much, I'm testing the fix now(on a linux box) and will push a new release in a few minutes.

Thanks!

Fabi0San, 2020-06-25:
https://github.com/Fabi0San/DuCalibrator/commit/12c41eaa14c305709e5fdc4892709b0f92bc4c57 should fix it.

closed 2020-06-25 by Fabi0San

Fabi0San, 2020-06-25:
The new release is out. You can update by using Octopi's "Software Update" option in the settings menu.
You may have to force check for updates in the advanced section.
Thank you @BrunoBrutus @Danbam and @snown for flagging this.

BrunoBrutus, 2020-06-25:
Thank yoy @Fabi0San all is working now :)