Fabi0San/DuCalibrator #3: Crashes into bed

azreal1983, 2020-06-25:
I am using a Anycubic delta Kossel. The auto calibrate works with the probe and it is set as a NC (Normally closed) probe in Marlin,

When clicking probe it just lowers and hits the bed (when probe is attached)

azreal1983, 2020-06-25:
This is before the first probe. it doesn't register the first probe point

azreal1983, 2020-06-25:
I think you have written this for a bltouch? as it is asking me to stow and deploy probe at each probe point. Which is currently dissabled in my marlin config?

Fabi0San, 2020-06-25:
>I think you have written this for a bltouch? 

No not at all, it is really probe agnostic.
It just moves to Z[SafeHeight] and perform a G30.
Your Marlin may be configured to ask for manual probe deployment.
I suggest you get the G30 working properly first.

azreal1983, 2020-06-25:
Ahh I see. But the G30 is a single probe point command, which will request as per the G30 for the probe to be deployed for each point. The only way round this is the remove the #define PAUSE_BEFORE_DEPLOY_STOW

Danbam, 2020-06-26:
does that need to be done in the firmware?

Fabi0San, 2020-06-26:
Yes, but make sure whatever you are doing is safe.
What kind of probe do you have, do you have to manually deploy it?

Danbam, 2020-06-26:
yes it has to be manually deployed. It is the factory probe on an Anycubic delta linear plus

Fabi0San, 2020-06-26:
I see, yes you may want to disable that the way @azreal1983 did, just keep in mind you have to remember to deploy the probe before starting calibration, also make sure the SafeHeight configuration leaves enough space for the probe to fit under the nozzle.

Fabi0San, 2020-06-26:
I may be able to make this easier with the E switch on G30, I'll look into it.

Danbam, 2020-06-26:
Working now and on my 8th cycle of probing. Hope it will be dialled in well soon :-)

closed 2020-06-28 by Fabi0San

Fabi0San, 2020-06-28:
With the v1.0.4 release you no longer need to disable PAUSE_BEFORE_DEPLOY_STOW.
It should now pause on the first point and last point of a trial to deploy/stow the probe, operate it as you normally would for probing operations.
Make sure safe height setting can accommodate the probe under the hot end.