Fabi0San/DuCalibrator #5: After first Apply

azreal1983, 2020-06-25:
After first apply calibration I cannot The fetch geometry fails to get anything. If referesh to broswer i get 
[image]

Fabi0San, 2020-06-25:
On Chrome press F12 to see the developer pane where you will find the console, where error messages may give us a clue of what went wrong, also check the terminal tab, it seems your printer didn't like the new settings.

azreal1983, 2020-06-25:
I was in firefox. I had cleanup plugin and start over.
The current geometry vanished. But Also the simulation values of 400 for steps were written to my EEPROM. Thus the printer was flying as my step are usually 80

azreal1983, 2020-06-25:
[image]

Fabi0San, 2020-06-25:
That is very weird, did you switch the firmware setting from simulation to marlin after doing fetch and probe?
The "Console" tab is the one with the interesting info you got a snapshot of "Elements"

azreal1983, 2020-06-25:
nope didn't switch. but all the current geometry were blank, so I just though it was rendering wrong, and apply calibration

azreal1983, 2020-06-25:
[image]

azreal1983, 2020-06-25:
should I cleanup the plugin again? Seems to not like loading the settings after the first run.

Fabi0San, 2020-06-25:
It is not connected to your printer, I can clean up some of the errors by checking for that but your printer doesn't seem to be connected to Octoprint.

azreal1983, 2020-06-25:
[image]
[image]
Connected but still the same

BrunoBrutus, 2020-06-25:
Hello @azreal1983, did update to the latest release 1.0.2?
I was having similar errors with the first one.
On my side I had to force the update check.

azreal1983, 2020-06-25:
DuCalibrator (1.0.2) that's the version I am on

Fabi0San, 2020-06-25:
I think I got it, trying to find out a repro.
You can try to work around by setting it to simulation, refresh the browser, connect the printer, then back to marlin setting, dont refresh and it may be functional then.

Fabi0San, 2020-06-25:
What version of OctoPrint are you on?

azreal1983, 2020-06-25:
OctoPrint 1.4.0

Fabi0San, 2020-06-25:
I cannot reproduce the scenario where the plugin is bound to controls before getting the printer status, but I'll add a check for that so we don't fail initialization.
Meanwhile you can try my workaround of running init(browser refresh) while firmware is set to simulation, and switch to marlin before using it. 
That is assuming you can change the setting while in this broken state.
Fix coming out later today.

azreal1983, 2020-06-25:
I tried it didn't work.
I don't mind waiting. If this works as expected it will be a super plugin. (and will get a lot of downloads on the delta Kossel group). 
I have made using a jig some new carbon rods so they are very closely sized. 
If this can fully calibrate my radius and offsets I will use this over Marlins G33 P10

Fabi0San, 2020-06-25:
I have been using this for a year now, it is huge help.
Truth is we shouldn't be calibrating that often given arms don't grow and endstops don't move.
Turns out the belts are the ones stretching and contracting and no one accounts for them.
My routine calibration now is to set sizes in stone and only calibrate steps/mm using this plugin as they change every day with temp and humidity!

Fabi0San, 2020-06-25:
Goes without saying, other methods can calibrate a belt stretch by changing radius and rod size, but that throws off dimensional accuracy and make for some oval looking circles.

azreal1983, 2020-06-25:
Have you changed your belts to some better stiffer ones e.g. gates?

Fabi0San, 2020-06-25:
Yes, I've tried it all, keep in mind that I'm dealing with micron level stuff here.

I found that the different belts with different core materials had different spring stiffness behavior which affects ringing on high acceleration scenarios but all of them had also their own thermal expansion coefficient which is never 0.
If anything, springier belts did better because you can tension them a lot and the springiness would compensate some of the thermal expansion, while "static" core belts would just expand loose tension.

Add to that the belt sprocket eccentricity which for gates is specified at 3 thous(75 microns) and you see how belt drives are pure evil :)

I'm way on the deep end here, none of this matters much for 3d printers doing .1mm / layer and 80 steps per mm. ;)

closed 2020-06-26 by Fabi0San

Fabi0San, 2020-06-26:
1.0.3 has just been released and has the fix for this but I have a suspicion that printer messages are not being propagated to all plugins and further messages wont come by, please let me know if something weird happens, attach the logs from the browser's console output on the developer tools window.