Fabi0San/DuCalibrator #6: ROD LENGTH corrections aren't fetched.

BrunoBrutus, 2020-06-25:
Hello @Fabi0San 

Have one possible enhancement request:
I have added on my delta some rod length corrections to calibrate the print size according to each tower direction, exemple:
_**#define DELTA_DIAGONAL_ROD_TRIM_TOWER {2.391, -0.024, -2.367}**_
Unfortunately, that is not possible to get from the firmware.
Your calculation, appears to, uses those values to calibrate.
Do you think it's possible to add those values manually to the initial fetched data?

Thank you

Fabi0San, 2020-06-25:
I have a Marlin PR ready to go out so it will support setting and fetching these via M665.
Until then, keep that unchecked.(Though yes major improvements come from that as no one can build arms with same size up to the micron.

Fabi0San, 2020-06-28:
Merged to Marlin bugfix branch

BrunoBrutus, 2020-06-28:
Thank you @Fabi0San for your efforts.
Unfortunately, when I moved to the latest MArlin 2 Bugfix I got major layer shifts.
Already made several tests (borrow a 32bits board, voltages, timmings,...) and appears to be related to my TMC2208 V1.0 drivers from Fysect.
So, got stuck with Marlin 1.1.9.1 until I can afford to buy new drivers, or a new board.
That's why the possibility to have those values available for manual edition would be great.
Anyway, I'm OKAY to close this request :)

Thank you

closed 2020-06-28 by Fabi0San