Fabi0San/DuCalibrator #8: M112: Emergency Stop

Danbam, 2020-06-26:
I have just updated the software and can now fetch the current geometry. Once this is done, when I klick on probe bed and fit the probe I get a M112 error requiring printer reset.

Fabi0San, 2020-06-26:
Can we get the last few lines from the terminal tab before the printer stopped?
Is your configured height int the ballpark of what it should be? 
Marlin freaks out when G30 cant find a bed close to where it should be, assume probe malfunction and stop to avoid a bad crash.

Danbam, 2020-06-26:
here you go. Regarding bed height, the z offset os around -20.00 so it could be the case, but delta calibration works

Recv: echo:busy: paused for user
Recv:  T:17.68 /0.00 B:18.21 /0.00 @:0 B@:0
Recv: echo:busy: paused for user
Recv:  T:17.68 /0.00 B:18.21 /0.00 @:0 B@:0
Recv: Stow Z-Probe
Recv: echo:busy: paused for user
Recv:  T:17.68 /0.00 B:18.39 /0.00 @:0 B@:0
Recv: echo:busy: paused for user
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

Fabi0San, 2020-06-26:
Oh we have seen this before, your probe requires manual intervention on each probing point, you might want to disable that.
Check issue https://github.com/Fabi0San/DuCalibrator/issues/3 and let me know if that is your problem.
In the end your printer should be able to run a G30 from the configured SafeHeight without intervention or error messages.

closed 2020-06-26 by Fabi0San