Fabi0San/DuCalibrator #9: Probe number

azreal1983, 2020-06-26:
I saw from your readme about the number of probes
50 is good 500 is slow etc - I think this is relating to CPU speed. I do not have this limitation. 

Do you have guidelines on diminishing return for the number of points?

I was thinking is it best to start with less e.g. 50 and when the RMS gets below a value increase to increase the number of points

azreal1983, 2020-06-26:
currently on trail 16 RMS 0.036 Min 0.080 Max 0.060, how long should I do the calibrate and probe loop?

Fabi0San, 2020-06-26:
Very good. I'm glad it is working out for you.
How many probings really depend on the repeatability of your machine, may be 3, may be 30.
Take a look a the history on the bottom you'll be able to see when RMS stalled.

To improve on that you need repeatable endstops and probes. 
If you have TMC drivers take a look at TMC_HOME_PHASE, I added it marlin bugfix branch a few months ago.
It is also very important to increase marlin precision via SERIAL_FLOAT_PRECISION, I added that to marlin 3 days ago. 
Set it to 5.

As for the probe points, it is vital that the probing has a higher frequency than your smallest artifact, otherwise it can be misrepresented. 
For instance I can only see the GT2 gear eccentricity artifact at 500+ points. 
But that number will vary printer to printer, hard to pick a number that will fit all.

azreal1983, 2020-06-26:
Where is TMC_HOME_PHASE in my firmware, couldnt find in Config.h or Config_adv.h

Fabi0San, 2020-06-26:
It is in _adv but only in the bugfix branch of Marlin:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h

azreal1983, 2020-06-26:
[image]

I have increased to 600 probe points now. 

From the screenshot, my bed is not 100% level with the frame, does that matter for the calculation as the bed is glass. so it is flat.

Fabi0San, 2020-06-26:
Yes, we assume the glass is flat, but the machine's geometry errors will make its computed plane not parallel to the bed.
The blue disk is the bed, the red dots are the computed coordinate where it touched the bed, the delta is your geometry error.
Being not leveled denotes a endstop trim error, it is really hard to get this right if your endstops don't have good repeatability.

Try fetching and probing a few times without changing anything(without clicking apply) the results should be the same, the difference between the results will show how repeatable your setup is. 
You cannot calibrate it past your machine's repeatability.

Fabi0San, 2020-06-26:
I think you are pushing this too far, it should have converged in 5 trials or so. If you see RMS go up from previous trial you probably went too far ;) Work on repeatability and precision, it would be a better use of your time ;)

azreal1983, 2020-06-27:
What do you mean to far? As in too many points ?

I also disabled changing my rod length as I know this length quite well (do within + or - 0.3mm. Perhaps for future max change can be added to some fields?

Fabi0San, 2020-06-28:
Too many trials. more than 5 is overkill.

Rod and radius varying far from reasonable indicates your steps/mm are likely wrong.
I suggest measuring it with a machinist dial.
I plan to introduce a different probing routine with 1-2-3 block to disambiguate that. 
Until then your first calibration of steps/mm have to be manual.

closed 2020-06-30 by Fabi0San