Fabi0San/DuCalibrator #18: question

azreal1983, 2021-01-30:
I built brand new rods using a jig. and I am pretty confident they are all 272mm long

With the algroythm to calculate the correct gemoetry, is it best to leave rod length unticked and just calibrate the towers and the endstop offsets?

Also you mention calculating the steps per unit manually. - any advice for doing this ona  delta? I have seen the tri block shape which run parallel to each tower. is this the best option?

Fabi0San, 2021-01-30:
I'm yet to implement limits on the adjustments, your rods are certainly off by at least a few microns. 
You should leave that unticked for the first few iterations. You may be able to calibrate the steps/unit using the sw if the rod lengths are fixed at 272. 
I don't know what is this tri block shape. I use a machinist dial on the carriages.

azreal1983, 2021-01-30:
thank you for you reply. I have calipers and a micrometer. How woudl you calibrate the steps/unit do you know of a guide?